rgerganov/footswitch #11: configured semicolon, got ñ

wolfwolker, 2015-01-25:
Firs of all, footswitch is awesome.

But I'm trying to configure the pedal to write semicolons using `footswitch -k ';'` or  `footswitch -k \;` or `footswitch -S 0x33`, then output of `footswitch -r` is fine, so everything seems well configured, but when I press the pedal it prints ñ

rgerganov, 2015-01-26:
Thanks for the bug report. Did you by any chance try to program a semicolon with the original Windows program? If that works then the problem is clearly in footswitch and I will fix it. I don't have the pedals right now, so I cannot try it myself ...

wolfwolker, 2015-01-27:
Fixed! you can close this bug now. The problem was me hehe. In my keyboard distribution semicolon has no key, need a key combination shift+comma, I configured the pedal with this combination and voila, works fine!

closed 2015-01-27 by wolfwolker