rgerganov/footswitch #12: Any way of implementing 'onRelease'?

sandric, 2015-03-29:
Isn't there any way to programmatically emulate releasing of pedal? I mean polling with some delay? thx

timofurrer, 2017-04-28:
That would be great! in X11 a KeyPress and KeyRelease is omitted...

rgerganov, 2017-04-28:
Unfortunately the hardware doesn't support this (no such events are being sent). If someone comes up with some clever hacks how to emulate this, I'd be happy to accept it.

timofurrer, 2017-04-28:
I check if I can come up with a xbk hack.

hellmind, 2023-11-04:
> Unfortunately the hardware doesn't support this (no such events are being sent). If someone comes up with some clever hacks how to emulate this, I'd be happy to accept it.

How can the hardware not support it when using modifier keys like Shift, Control, Alt, and Win actually sends the release event?

mmarshall540, 2024-01-25:
> How can the hardware not support it when using modifier keys like Shift, Control, Alt, and Win actually sends the release event?

It sends the release event for the same key.  `footswitch` is only programs the pedal.  If the pedal doesn't know how to do what you want, it won't be able to.

You can do it in software, but it's software that stays running on your computer (as opposed to just programming the foot pedal).

See [the `tap-macro-release` function included in Kmonad](https://github.com/kmonad/kmonad/blob/eea2585fbb3e36f4211b8e2bf0da5d42ec35050f/keymap/tutorial.kbd#L517).