rgerganov/footswitch #17: cannot find footswitch with VID:PID=c45:7403 [init(), footswitch.c:96]

torsava, 2016-04-26:
I believe I do have a compatible foot switch (though I bought it off of AliExpress):

    $ lsusb -d 0c45:7403
    Bus 002 Device 010: ID 0c45:7403 Microdia Foot Switch

However, when I try to use the utility I get the aforementioned message:

    $ footswitch -r
    cannot find footswitch with VID:PID=c45:7403 [init(), footswitch.c:96]

I am on Fedora Linux 23, fully updated current system. I do have the [hidapi and hidapi-devel](https://admin.fedoraproject.org/pkgdb/package/rpms/hidapi/) packages installed. I followed the installation procedure.

I even tried to modify the C code to investigate if it correctly understands it runs on Linux and not OSX (footswitch.c line 76) and it does. Now I'm out of ideas.

Any help or pointer would be greatly appreciated!

torsava, 2016-04-26:
Solved, I was running the command without sudo / not as the superuser. With sudo it works!

closed 2016-04-26 by torsava

tanin47, 2016-09-23:
I have this exact problem, and running it with sudo solves it.