rgerganov/footswitch #19: OSX 10.11.3 install error

tranlm, 2016-05-03:
Hi there,

I think my version of OS X is preventing installation due to increased security. Here's the error I'm getting:

```
➜  Downloads git clone https://github.com/rgerganov/footswitch.git
Cloning into 'footswitch'...
remote: Counting objects: 78, done.
remote: Total 78 (delta 0), reused 0 (delta 0), pack-reused 78
Unpacking objects: 100% (78/78), done.
Checking connectivity... done.
➜  Downloads cd footswitch 
➜  footswitch git:(master) make
cc footswitch.c common.c debug.c -o footswitch -Wall -DOSX -lhidapi
➜  footswitch git:(master) ✗ sudo make install
Password:
/usr/bin/install -c footswitch /usr/bin
install: /usr/bin/footswitch: Operation not permitted
make: *** [install] Error 71

```
Is it possible just to change the install location in the makefile? Or would I have to do something more extensive?

Thanks!

rgerganov, 2016-05-05:
Hi, yes, you can change the install location. I am not sure why it is not possible to install in /usr/bin on newer versions of OSX.

tranlm, 2016-05-24:
It worked. Thanks a lot!

closed 2016-09-07 by rgerganov