rgerganov/footswitch #20: Files missing when installing from command line

jamminjames, 2016-07-01:
Following the command line installation instructions, I get the following error when I get to 
$ make

cc footswitch.c common.c debug.c -o footswitch -Wall `pkg-config hidapi-libusb --libs`
footswitch.c:23:19: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
                   ^
compilation terminated.
common.c:23:19: fatal error: ctype.h: No such file or directory
 #include <ctype.h>
                   ^
compilation terminated.
debug.c:22:19: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
                   ^
compilation terminated.
make: *** [footswitch] Error 1

There seem to be missing files, how can I get them? Would really like to get this foot pedal working. Thanks!

rgerganov, 2016-07-01:
Looks like you don't have a C compiler installed.  If you are using Ubuntu, you can try this:

$ sudo apt-get install build-essential

jamminjames, 2016-07-01:
Well, I ran:
 $ sudo apt-get install pkg-config libusb-dev
...and then the 'make' command worked. But now if I use the 'footswitch -r' command, for example, I'm getting:

cannot find footswitch with VID:PID=c45:7403 [init(), footswitch.c:94]

... the address for the footswitch should be 0c45:7403, though, how is it getting c45:7403, and how can I fix that?

rgerganov, 2016-07-01:
You need to either use 'sudo footswitch ...' or install the udev rules with 'sudo make install'

jamminjames, 2016-07-01:
okay, i entered 'sudo make install', but still had to use sudo in the footswitch command anyway. Would I need to uninstall & reinstall to avoid using sudo everytime?

Anyway, at least I can get it to work now. Thank you very much!

jamminjames, 2016-07-01:
How do you find the switches needed to program certain keystrokes, for example, I'd like a pedal to enter: arrow down; double click.

jamminjames, 2016-07-01:
Found the hex code for arrow down, but can't find anything for touchpad double-click. unfortunately, just using ENTER won't work in this application.

jamminjames, 2016-07-01:
The equivalent of right-click, open would work too.