rgerganov/footswitch #21: hidapi/hidapi.h file not found

marcellkiss, 2016-08-24:
I just installed the 0.8.0 release candidate of hidapi using brew. When I run `make` in the footswitch folder I get the following error:

```
cc footswitch.c common.c debug.c -o footswitch -Wall -DOSX -lhidapi
footswitch.c:28:10: fatal error: 'hidapi/hidapi.h' file not found
#include <hidapi/hidapi.h>
         ^
1 error generated.
make: *** [footswitch] Error 1
```

I guess, somehow the hidapi install is not ok, but I don't have any idea what I could change.. 

Could you help me out? 

Thanks in advance!

rgerganov, 2016-08-24:
Hm, this is weird because I am using hidapi 0.8.0-rc1 on my Mac without problems.  I am also using brew and hidapi is installed in /usr/local/Cellar/hidapi/0.8.0-rc1.  Could you paste the output of the following commands:

```
$ pkg-config --cflags hidapi
$ ls -l /usr/local/include/hidapi

```

marcellkiss, 2016-08-24:
Wow, thanks for the fast answer, the output is:
-I/usr/local/Cellar/hidapi/0.8.0-rc1/include/hidapi

So it looks like it's at the right place, right?

rgerganov, 2016-08-24:
Do you have a symlink /usr/local/include/hidapi which points to /usr/local/Cellar/hidapi/0.8.0-rc1/include/hidapi?

marcellkiss, 2016-08-24:
Yes, it looks like this:
`lrwxr-xr-x  1 marcellkiss  admin  41 Aug 23 12:55 /usr/local/include/hidapi -> ../Cellar/hidapi/0.8.0-rc1/include/hidapi`

rgerganov, 2016-08-24:
I patched the Makefile to put explicit include paths on OSX:  https://gist.github.com/rgerganov/840c2780e948ecaab001952e11d5b7c8

Could you try it out and let me know if it works for you?

marcellkiss, 2016-08-25:
Thanks! What should be the value of CC?
ᐧ

rgerganov, 2016-08-25:
This is the C compiler, it should be set by default.  What is the output with the modified Makefile?

marcellkiss, 2016-08-25:
`Makefile:20: *** missing separator.  Stop.`

I couldn't figure it out yet : /
ᐧ

rgerganov, 2016-08-25:
Make sure that you are using tabs, not spaces for indentation.  If you download the raw file from my gist, it should be fine.

marcellkiss, 2016-08-25:
Hm, previous problem solved, but it throws the missing hidapi.h error
message like before:

cc footswitch.c common.c debug.c -o footswitch -Wall -DOSX `pkg-config
--cflags hidapi` `pkg-config --libs hidapi`

footswitch.c:28:10: fatal error: 'hidapi/hidapi.h' file not found

#include <hidapi/hidapi.h>

         ^

1 error generated.

make: *** [footswitch] Error 1
ᐧ

rgerganov, 2016-08-25:
OK, let's try one more change: replace the following include in `footswitch.c`

`#include <hidapi/hidapi.h>`

with:

`#include <hidapi.h>`

and try again with the modified Makefile

marcellkiss, 2016-08-25:
Now it looks quite good, I can run it locally `./footswitch`. It's not so
important, but I can't install it globally:

➜  footswitch git:(master) ✗ sudo make install

Password:

/usr/bin/install -c footswitch /usr/bin

install: /usr/bin/footswitch: Operation not permitted

make: *** [install] Error 71

➜  footswitch git:(master) ✗

And another important question, how can I represent buttons like:
 - Command
 - Option
 - Space

Thanks, hopefully this was my last question. Your helpfulness is amazing,
really, thanks a lot! Keep up the good work!!
ᐧ

rgerganov, 2016-08-25:
No problem at all, thanks for the bug report, I will commit the fixes soon.
- the problem with `make install` has been reported (issue#19) from another user as well, try to change `/usr/bin` in  the Makefile with some other dir that is in your `PATH` and run `make install` again
- for the `space` key, `footswitch -k space` should do the job. Need to check for `Command` and `Option`

closed 2016-09-05 by rgerganov