rgerganov/footswitch #26: [Linux Mint] Footswitch seems to work once

SeanZicari, 2017-03-15:
Greetings. I love the software and actually bought a footswitch because I found this software. Thanks for writing it!

I'm having an issue where the foot press registers once, maybe, and then doesn't work again. I'm limited in my ability to troubleshoot and am wondering if anyone has any suggestions on figuring out why the footswitch doesn't work 100% of the time.

rgerganov, 2017-03-16:
Hi! Could you paste the output of the `lsusb` command when the switch is plugged in? What key sequence did you program the switch with? Do you see the same behavior when you plug the switch in another box?

SeanZicari, 2017-03-17:
I haven't tested on another machine yet. I have the switch programmed to `KP_Divide`, which I have set as a custom key in Linux Mint to toggle a script.

`lsusb` output:

```bash
Bus 001 Device 002: ID 046d:082d Logitech, Inc. HD Pro Webcam C920
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 045e:0040 Microsoft Corp. Wheel Mouse Optical
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 0c45:7404 Microdia
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

`footswitch -r` output:

```bash
[switch 1]: unconfigured
[switch 2]: <KP_Divide>
[switch 3]: unconfigured
```

SeanZicari, 2017-03-17:
I should add - usually the switch works on the first press but not after that.

SeanZicari, 2017-03-20:
It seems like between the footswitch and the Synergy software I'm running, keyboard focus is completely stolen after I press the footswitch once. I get a gray dotted line around choices that I click on, and I have to click twice to get my click to register.

rgerganov, 2017-03-22:
Hm, I am not sure what is wrong. Maybe try to isolate the problem and see if it occurs without Synergy running or using another key. Sorry for not being helpful!

SeanZicari, 2017-03-22:
The software itself is very helpful - just checking to see if there were any known issues. I'll keep experimenting and get back to you.

bzg, 2017-07-28:
Hi,

I've been hit by the same issue.  I use footswitch like this:

`~$ footswitch -1 -k up -2 -k insert -3 -k down`

Hitting the _right_ pedal sends `down` (`^[[B`) once, and then repeatedly, as if my pedal was continuously pressed.  Pressing another pedal stops the sending of events, but then the right pedal is not sending anything anymore.

Unplugging and re-plugging the footswitch makes the right pedal available again, but just once.

I suspect the problem might be mechanical since it has been working fine for a while, but how can I test it further?

SeanZicari, 2017-08-30:
I can vouch for the repetitive behavior. I am using the pedal to try and toggle a Google Hangout's audio (the pedal is programmed to send Ctrl-d). When I hit it once, the Hangout rapidly mutes and unmutes.

bzg, 2017-08-30:
Thanks @SeanZicari for confirming this weird behavior. I'm wondering if this is a software or a hardware issue, I don't know how to test this.

x1101, 2018-04-03:
Confirming the issue that @bzg and @SeanZicari are reporting. I bought my pedal (the `0c45:7404`) to use as a push-to-talk key. Worked fine for several sessions, and then the behavior described started. Once plugged in, it works *once* and then continues to spam the key until unplugged.

I'm happy to provide any assistance troubleshooting, if needed.

krasiyan, 2018-06-29:
@SeanZicari @bzg @x1101 I had the same issue with my pedal - a few seconds after it was plugged in, a constant or a single key press would be registered (no matter if I actually pressed it or not). Nothing apart from that key press would show up on `xev` / `showkey`. My pedal worked just fine initially and the issue appeared after not using it for several months.

My issue turned out to be a hardware one - apparently the IR diode stopped making good contact with the PCB which would cause the switch to register itself as pressed down all the time. I managed to verify that by flashing my cellphone flashlight in the photo-resistor which would register a "release". As a temporary fix I just carefully bended both diodes inwards so they would make a better electrical contact:

[image]

This is the switch I am using, so other models might have different inner-workings (rather doubtful)
[image]

I also managed to reproduce the problematic behavior after the fix - I simply held the switch pressed and plugged it in :D

P.S. @rgerganov If anyone else confirms the same hardware issue maybe we can add a note in the `README` about it?

rgerganov, 2018-06-29:
@krasiyan thanks a lot for sharing this, great troubleshooting and fix :) I will certainly add this to the README

rgerganov, 2018-07-10:
The README is updated, so closing this for now.

closed 2018-07-10 by rgerganov

x1101, 2018-07-19:
Confirmed. I also bent the two sensors inward and my footswitch works once more. Big thanks ot @krasiyan for providing such a simple fix!

ianwillie, 2018-08-19:
I have had the problem mentioned above with a FS3-P. I re-soldered the connections of the IR emitter and the receptor and the problem persisted. When I bent these slightly as suggested in this post it started to work perfectly. 
I think the problem is related to the angles and one or other or both have to tilted slightly downward  for the IR radiation to be picked up properly by the detector.

tricoos, 2020-07-18:
@krasiyan  @ianwillie  @x1101 Thanks for the info, I have the same problem with the FS3-P (https://pcsensor.com/usb-foot-control-keyboard-mouse-button-three-switch-pedal-fs3_p-p180.html) which has three pedals. How did you open yours without destroying it?

tricoos, 2020-12-28:
Just for everyone else as I damaged one of my pedals when opening it: you have to remove the three screws at the bottom, only than can you push out the metal bars on any of the switches from the side. Then you can remove the plastic cover for the pedal. Do not try to push open any pedal with force as this will break the holder at the bottom.

Stoatwblr, 2023-07-06:
> Confirmed. I also bent the two sensors inward and my footswitch works once more. Big thanks ot @krasiyan for providing such a simple fix!

I'll add my vote here

Checking with a smartphone camera will show if the IR leds are emitting. 
You can test the IR photodiode by saturating it with a red laser pointer (if the output is triggering, the laser will shut it down)

The solder joints on my 3 pedal unit looked bad but resoldering made no difference. That's when I checked using the method suggested above

Tilting the led/photodiode pair inwards by about 10-15 degrees each makes things _much_ more reliable (I was getting false triggering with less tilt) and using the laser allows more reliable testing of the angles required to get a reliable optical path

I don't know why the makers didn't use a standard photo interrupter package. These are cheap and already aligned