rgerganov/footswitch #27: How to avoid using sudo for footswitch everytime?

zenny, 2017-05-16:
Hi,

Installed _footswitch_ with '**sudo** make install' for the device ID 0c45:740**4** as of below. 

But running footswitch requires superuser everytime. How to avoid 'sudo' for running _footswitch_ every time?!  Thanks. 

> $ lsusb -d 0c45:7404
> Bus 001 Device 003: ID 0c45:7404 Microdia
> 
> $ make
> cc footswitch.c common.c debug.c -o footswitch -Wall -I/usr/include/hidapi   -lhidapi-libusb  
> 
> $ sudo make install
> /usr/bin/install -c footswitch /usr/local/bin
> /usr/bin/install -c -m 644 19-footswitch.rules /etc/udev/rules.d
> 
> $ footswitch -r
> Cannot find footswitch with VID:PID=c45:7403 or VID:PID=c45:7404 [init(), footswitch.c:101]
> 
> $ sudo footswitch -r
> [sudo] password for zenny: 
> [switch 1]: a
> [switch 2]: b
> [switch 3]: c

Cheers,
/z

rgerganov, 2017-05-16:
Could you paste the output of `lsusb -v -d 0c45:7404`?

zenny, 2017-05-16:
> $ lsusb -v -d 0c45:7404
> 
> Bus 001 Device 003: ID 0c45:7404 Microdia 
> Couldn't open device, some information will be missing
> Device Descriptor:
>   bLength                18
>   bDescriptorType         1
>   bcdUSB               2.00
>   bDeviceClass            0 (Defined at Interface level)
>   bDeviceSubClass         0 
>   bDeviceProtocol         0 
>   bMaxPacketSize0         8
>   idVendor           0x0c45 Microdia
>   idProduct          0x7404 
>   bcdDevice            0.01
>   iManufacturer           1 
>   iProduct                2 
>   iSerial                 0 
>   bNumConfigurations      1
>   Configuration Descriptor:
>     bLength                 9
>     bDescriptorType         2
>     wTotalLength           59
>     bNumInterfaces          2
>     bConfigurationValue     1
>     iConfiguration          0 
>     bmAttributes         0xa0
>       (Bus Powered)
>       Remote Wakeup
>     MaxPower              100mA
>     Interface Descriptor:
>       bLength                 9
>       bDescriptorType         4
>       bInterfaceNumber        0
>       bAlternateSetting       0
>       bNumEndpoints           1
>       bInterfaceClass         3 Human Interface Device
>       bInterfaceSubClass      1 Boot Interface Subclass
>       bInterfaceProtocol      1 Keyboard
>       iInterface              0 
>         HID Device Descriptor:
>           bLength                 9
>           bDescriptorType        33
>           bcdHID               1.10
>           bCountryCode            0 Not supported
>           bNumDescriptors         1
>           bDescriptorType        34 Report
>           wDescriptorLength     260
>          Report Descriptors: 
>            ** UNAVAILABLE **
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x81  EP 1 IN
>         bmAttributes            3
>           Transfer Type            Interrupt
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0008  1x 8 bytes
>         bInterval              10
>     Interface Descriptor:
>       bLength                 9
>       bDescriptorType         4
>       bInterfaceNumber        1
>       bAlternateSetting       0
>       bNumEndpoints           1
>       bInterfaceClass         3 Human Interface Device
>       bInterfaceSubClass      1 Boot Interface Subclass
>       bInterfaceProtocol      2 Mouse
>       iInterface              0 
>         HID Device Descriptor:
>           bLength                 9
>           bDescriptorType        33
>           bcdHID               1.10
>           bCountryCode            0 Not supported
>           bNumDescriptors         1
>           bDescriptorType        34 Report
>           wDescriptorLength      41
>          Report Descriptors: 
>            ** UNAVAILABLE **
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x82  EP 2 IN
>         bmAttributes            3
>           Transfer Type            Interrupt
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0008  1x 8 bytes
>         bInterval              10
>

rgerganov, 2017-05-16:
Could you try to update the first line of 19-footswitch.rules to be:

`ATTR{product}=="*FootSwitch*"\`

and try again?

zenny, 2017-05-16:
> Could you try to update the first line of 19-footswitch.rules to be:
> 
> ATTR{product}=="*FootSwitch*"\
> 
> and try again?

No changes.

rgerganov, 2017-05-16:
You put an asterisk before and after FootSwitch, right?
If it doesn't work, you may try to use another attribute (like productId or vendorId) in the udev rule. I will also experiment later because I don't have the device right now.

zenny, 2017-05-16:
> You put an asterisk before and after FootSwitch, right?

Yep as of below:

> $ grep -Rn ATTR /etc/udev/rules.d/19-footswitch.rules 
> 1:ATTR{product}=="*FootSwitch*"\

zenny, 2017-05-16:
Hi again,

Replacing /etc/udev/rules.d/19-footswitch.rules with the following let run footswitch without sudo:

`KERNEL=="hidraw9[0-11]*", SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="0C45", ATTRS{idProduct}=="7404", SYMLINK+="footswitch"`

Let me know if there is an easier way than this. Thanks!

Cheers,
/z

zenny, 2017-05-17:
UPDATE:  The above stopped working after a reboot, thus back to square one, forcing to use sudo with footswitch.

zenny, 2017-05-29:
@rgerganov Did you have time to look at this issue? Appreciate that!

rgerganov, 2017-05-29:
@zenny I cannot reproduce this problem with my device and my linux distro (Ubuntu). What linux distribution do you use?

robrussell, 2018-03-28:
I had the same problem. Just tried the recommended change to 

> *FootSwitch*

 and it worked for me.

So now my /etc/udev/rules.d/19-footswitch.rules contains
```
ATTR{product}=="*FootSwitch*"\
MODE:="0666"\
GROUP="hid"\
RUN+="keymap $name 0x70066 screenlock"\
SYMLINK+="footswitch"

```

rgerganov, 2018-03-28:
@robrussell Thanks for the update. Could you paste the output of `sudo lsusb -v` so we can see the value of the `product` attribute of your device?

hagemt, 2018-03-30:
A fast hack (important to understand what it does) is as follows:

```
sudo chown 0:0 footswitch
sudo chmod 6711 footswitch
```

This makes the executable always run as `root`, no matter who runs it.