rgerganov/footswitch #31: Not working with Raspberry Pi

mzm2lh, 2018-02-08:
I have a PCSensor 413D:2107 foot switch and the program is failing because dev == (nil) at footswitch.c 112.  Can you suggest remedy?

rgerganov, 2018-02-08:
Does it work when you start it as root?

mzm2lh, 2018-02-08:
'footswitch -r' works when run as root:
 [switch 1]: a
 [switch 2]: b
 [switch 3]: c

But 'footswitch -1 x' or 'foot switch -2 x' or 'foot switch -3 x' returns:
  "error writing data ((null)) [usb_write(), foot switch.c:151]"

The program was installed when logged in as a normal (non-root) user (pi).

rgerganov, 2018-02-09:
You need to specify number of pixels when using -x. Like this:

footswitch -1 x 10

mzm2lh, 2018-02-09:
Sorry, my bad; I am entering commands as per documentation, and as root user, like:
  sudo footswitch -1 -k x
or
  sudo footswitch -k x

But I always get:
  "error writing data ((null)) [usb_write(), footswitch.c:143"

(Now line 143 instead of 151 because I took out my inserted fprint statements)