rgerganov/footswitch #36: error getting feature report on linux

Mte90, 2018-05-11:
I am trying to use the scythe footswitch without success.
When I execute `scythe -r` I get that `error getting feature report ((null)) [read_pedals(), scythe.c:136]` running as root but I restarted udev with:
```
udevadm control --reload-rules
/etc/init.d/udev restart
```
ALso I saw that on debian the `hid` group for udev doesn't exist so I am using for tests plugdev that I used in the past with different udev rules for USB devices.

Mte90, 2018-05-11:
I made also a new udev rule only for the scythe executable for testing.

rgerganov, 2018-05-11:
When running the program as root udev rules shouldn't matter. Make sure that hidapi library is up to date.

Mte90, 2018-05-11:
Yes it is the latest https://packages.debian.org/sid/libhidapi-dev

Mte90, 2018-05-11:
Maybe in the udev rules I have to add the vendorid/productid and see what happen.

Mte90, 2018-05-11:
I compiled the testgui and I am getting an error too if I am execute as normal user but as root I don't get any errors and I see also the input.

[image]

Mte90, 2018-05-11:
I am trying with this udev:
```
ATTR{product}=="Scythe FootSwitch*"\
MODE:="0666"\
ATTRS{idVendor}=="0426"\
ATTRS{idProduct}=="3011"\
KERNEL=="hidraw*"\
SUBSYSTEMS=="usb"\
RUN+="keymap $name 0x70066 screenlock"\
SYMLINK+="scythe"```

Mte90, 2018-05-11:
I can see the input of the pedal on the console so I think that they work but there are problems on access from other software.

Mte90, 2018-05-11:
I see the output on console when I run `udevadm monitor -e` so is probably something on udev.

Mte90, 2018-05-11:
I think that `keymap` is a command that is not available on debian and also on fedora (https://github.com/rgerganov/footswitch/pull/18#issuecomment-215801975).

Mte90, 2018-05-11:
Looking on https://packages.ubuntu.com/search?searchon=contents&keywords=keymap&mode=exactfilename&suite=bionic&arch=any is part of a package that on debian exist https://packages.debian.org/sid/utils/fnfxd

It is a daemon but I cannot install because:
```
mag 11 15:03:43 KITT fnfxd[23279]: fatal error: Could not open /proc/acpi/toshiba/keys.
mag 11 15:03:43 KITT fnfxd[23279]: Please make sure that your kernel has enabled the Toshiba option in the ACPI section.
mag 11 15:03:43 KITT fnfxd[23279]: For more information read the documentation and/or http://fnfx.sf.net/index.php?section=doc#kernel.
mag 11 15:03:43 KITT fnfxd[23279]: FnFX Daemon v0.3 (c) 2003, 2004 Timo Hoenig <thoenig@nouse.net>
mag 11 15:03:43 KITT systemd[1]: fnfxd.service: Control process exited, code=exited status=1
mag 11 15:03:43 KITT systemd[1]: fnfxd.service: Failed with result 'exit-code'.
```

Mte90, 2018-05-11:
after a while I made a python script that do the same of footswitch and support schyte: https://github.com/Mte90/pydal

My plan is to add a configuration for every button etc like footswitch but this solution doesn't have the problem that I have with this and is using evdev that detect every pressure of buttons also as non-root.

closed 2019-07-12 by rgerganov