rgerganov/footswitch #40: Configuring other mouse buttons than Left and Right

0x5a17ed, 2018-07-15:
Is it possible to program the Scythe's PCsensor footswitch to send different mouse buttons than the already supported left and right mouse button? 

My mouse for example has a Left, Right, Middle, 4th and 5th mouse button and I would like to program the footswitch to send the code for a 6th mouse button so I can bind it to a completely different function.

Unfortunately the functionality at least is not in this programming software yet and I was wondering if it is even supported by the hardware and the communication protocol. Sadly I have no expertise regarding the hardware or the communication protocol.

rgerganov, 2018-07-17:
How do you differentiate mouse buttons from other key combinations? Why not program the footswitch to emit, for example, Alt+Shift+1 and bind this to the function you want?

0x5a17ed, 2018-07-20:
Thanks for that recommendation, I'm currently testing this out for my personal use case to enable Push-To-Talk in my voice chat client.

I still think it could be useful in the future to be able to program other mouse buttons as well though.