rgerganov/footswitch #47: How to use footswitch with ctrl:swapcaps?

kenhys, 2018-11-27:
# Problem

Without ctrl:swapcaps, `footswitch -m ctrl -k t` works as expected (Ctrl+T).
But with ctrl:swapcaps, it seems that it doesn't work as expected.(just `T` is printed)

# How to reproduce

1. setxkbmap -option ctrl:swapcaps
2. footswitch -m ctrl -k t
3. Press center footswitch 

# Expected

Even though with ctrl:swapcaps, there is a way to enable Ctrl+T.
It seems that combination with capslock and t is not allowed.
Do you know the way?

rgerganov, 2018-11-27:
Sorry, I don't know how to do this.

kenhys, 2018-12-10:
Thanks, @rgerganov .
I want to use Ctrl+T for prefix of tmux, and it is happly if footswitch works as a alternative.
After all, I found that it is good enough to change it Alt-T as a workaround (modify .tmux.conf) in my case. (it doesn't conflict with other shortcut keys and works with ctrl:swapcaps in my environment)

kenhys, 2018-12-10:
There is a workaround,  so I'll close.

closed 2018-12-10 by kenhys