rgerganov/footswitch #50: Problems using multiple modifiers

FriedrichFroebel, 2019-07-12:
I am currently trying to get my PCSensor foot switch to work using multiple modifier keys at once with my German keyboard layout on Ubuntu 18.04.

Basically I need <kbd>Alt Ctrl + 7</kbd> which inserts `{` using the regular keyboard. Trying to use either

* `./footswitch -S 'e9 e3' -s 7` (leading to `<XF86Eject><Super_L>7` with `-r`) or
* `./footswitch -m alt -m ctrl -k 7` (leading to `ctrl+alt+7`)

does not seem to work. In the first case, it will just print `7` and the second case does not seem to produce any output at all.

Trying to use `ß` as a key (to get a backslash) does not work as well as this character cannot be encoded: `Cannot encode key 'ß'`.

Is there anything I can do about this?

(By the way: Using the "official" software on Windows the `ssharp` character does not work, too, as the software inserts `]` instead. Configuring the foot switch on Windows and using the configured keys on Ubuntu does not work as well as the wrong output gets produced.)

rgerganov, 2019-07-12:
> I am currently trying to get my PCSensor foot switch to work using multiple modifier keys at once with my German keyboard layout on Ubuntu 18.04.
> 
> Basically I need Alt Ctrl + 7 which inserts `{` using the regular keyboard. Trying to use either
> 

I guess you have some additional input handler which "translates" Alt+Ctrl+7 to `{`. I get no output when I press this on my keyboard. Why not simply using `footswitch -k '{'`?

FriedrichFroebel, 2019-07-12:
German keyboard layouts have a dedicated <kbd>Alt Gr</kbd> switch and `{` is the third option of the key `7`. According to `xev` this key seems to be mapped to `keycode 108 (keysym 0xfe03, ISO_Level3_Shift)`.

`footswitch -k '{'` does not produce any output, `xev` says this writes `keycode 248 (keysym 0x0, NoSymbol)`.