rgerganov/footswitch #53: Support for devices with more than 3 switches

patrickkelso, 2020-01-14:
I recently grabbed a 6 button usb keyboard ( https://www.amazon.com.au/gp/product/B00X76DS80/ ) for some work and was delighted to see it's keycode in footswitch already. It defaults to outputting a b c d e f and works perfectly. I was hoping to reprogram using footswitch rather than booting to Windows and found the footswitch only supports 3 keys. Any chance of an update to support more keys?

```
sudo /usr/bin/footswitch -r
[switch 1]: a
[switch 2]: b
[switch 3]: c
```

````
$ lsusb -v -d 1a86:e026

Bus 001 Device 046: ID 1a86:e026 QinHeng Electronics 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0xe026 
  bcdDevice            0.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     212
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      41
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
```

rgerganov, 2020-01-15:
I can give it a try if someone borrows me the hardware. Also if anyone else want to contribute this functionality, patches are welcome.

patrickkelso, 2020-01-15:
Happy to donate. Easier than me trying to code. It's a nifty device for tinkering with automation.

rgerganov, 2020-01-16:
Great but I am not able to find a seller that ships this switch to Sofia, Bulgaria. Send me an email to rgerganov(at)gmail.com and we can think how to arrange this. Thank you for your support!

rgerganov, 2020-01-29:
I created a `six-buttons` branch with a simple change that I believe could work. Could you please give it a try:
```
$ git clone https://github.com/rgerganov/footswitch
$ git checkout six-buttons
$ make
$ ./footswitch -r
```
Let's try only reading the device for now. If that works, I will push more changes for writing.

patrickkelso, 2020-01-29:
Brilliant.

 $ sudo ./footswitch -r
[switch 1]: a
[switch 2]: b
[switch 3]: c
[switch 4]: d
[switch 5]: e
[switch 6]: f

rgerganov, 2020-01-29:
Great. I pushed one more change, do a git pull and try to program all of the six buttons.

patrickkelso, 2020-01-29:
Works a treat. You've made me a very happy man.

piegamesde, 2020-08-19:
I purchased [this](https://www.amazon.com/iKKEGOL-Control-Customized-Keyboard-Multimedia/dp/B07553KW32) foot switch, which interestingly also has the ID `1a86:e026`. It works out of the box, but is wrongly viewed as device with thee switches.

linuxmaster, 2020-08-27:
It is funny to say, but I have been trying to get connected with https://pcsensor.com/ but they ignored me ^__^ I said them about this project and this is a good moment to support this project for them. Maybe they need to hear only Chinese?