rgerganov/footswitch #58: Support for new pedal 5131:2019 Pedal?

stefmat10, 2020-08-05:
I just got a new pedal from AliExpress and the VID:PID is 5131:2019 so it does not find it!

I have tried adding it to the list of VIDS in the source code but it still does not work so any suggestions?

scizzorz, 2020-08-06:
I'm experiencing the same issue - I just got a new pedal from Amazon that's branded the FS1-P with the same VID:PID. I wasn't able to get it to work with PCSensor, and the software download link provided by the pack-in manual doesn't have a lot of flexibility.

rgerganov, 2020-08-07:
Can you post the Amazon link and the software download link?

scizzorz, 2020-08-07:
[Amazon](https://www.amazon.com/gp/product/B0098PLPOI) and [software](http://hottestdealever.com/Software/S43%20Software%202020.zip). The software seems... sketchy...

scizzorz, 2020-08-07:
Relevant line of `lsusb`:

```
Bus 001 Device 029: ID 5131:2019 GZIOT USBSW
```

lamialily, 2020-09-30:
Just got one of these footpedals myself and encountered this project and issue while looking into a way to reprogram it from Linux. Like scizzorz, I'm not about to blindly trust the CD-R they sent me with the drivers. A quick disassembly showed it's rather basic, with mine just being a basic microswitch foot pedal while the USB connector has all of the electronics (if you could call a CH551G and a couple of capacitors that...).

I found a project on hackaday that had an English datasheet for the CH551 and its sister chip, the CH552, that might lend some insight on how to reprogram it. I'll look into it in more detail myself to see if I can add the functionality to this software. Until then, here it is for your perusal. https://hackaday.io/project/169671-ch552-dragon

jasonm23, 2022-09-10:
A couple more vendor/prod id pairs:

| USB Device Name | Vendor ID | Product ID | 
|----:|:---:|:---:|
| Footswitch1F1 (RDing) | 3141 | 29299  |
| USBSW (GZIOT) | 20785 | 8217 |

Footswitch1F1 (RDing) - registers key down and key up (I got this one a long time ago, and programmed it, but I forgot how :sad:)

USBSW (GZIOT) - is just out the box and is set to <kbd>Return/Enter</kbd>. It sends down/up for each press down, nicer build quality though. I guess I need to pry this one open and see what's up.

jasonm23, 2022-09-10:
potentially interesting : https://www.codrey.com/learn/cheap-ch551g-mcu-with-usb-controller-part-1/

jasonm23, 2022-09-10:
FYI mine appears to have a CH552G

jasonm23, 2022-09-10:
https://github.com/MarsTechHAN/ch552tool

Izzard-UK, 2022-09-28:
Hello. I've just bought one of these pedals. I bought the cheapest option and I've run into the same issue.
It's a USBSW (GZIOT) | 0x5131 | 0x2019 … and I want to send different keypresses for DOWN and UP events, but unfortunately this only sends a key on DOWN.

I've been able to change the key that it sends from Enter to something else, using usbsw_config_en.exe - but it otherwise isn't compatible with PCSensor's configuration software for reasons which are now painfully obvious.

So.. has anyone figured out a way to make the GZIOT USBSW device function like the RDing device?  That would be awesome.

jasonm23, 2022-09-29:
@Izzard-UK - beyond building a new firmware, for now we're stuck.

The GZIOT firmware was programmed to send down/up from one hardware switch close.  As you've seen using the Win32 "official" software it's only possible to change which key it sends.

I set that to Esc, which can be used by Karabiner as a  way to trigger sticky modifiers or macros.

Izzard-UK, 2022-09-29:
That’s a shame. They’re not a bad little device for the money, and I can see use-cases for both types as described above.  The main issue is that marketplaces like Amazon, eBay and AliExpress are flooded with lots of rebrands / iterations of these pedals but the descriptions never make it clear which type you’re getting.

jasonm23, 2022-09-29:
I got hold of one for about $2.

Unfortunately when I opened it up there was no MCU in there, even the switch hadn't been soldered to the cable.

There are some cheap 3 pedal pcsensor units available on AliExpress and they are pretty good though and $20ish

Izzard-UK, 2022-09-29:
Can the better version actually be used to press two different keys? In other words, send one key code on down and a different one on up?

jasonm23, 2022-09-29:
Yeah, that works, key down/up is registered.  The software is better and works on mac,windows,linux too. (PCSensor)

Mind you, I wanted to use them as modifiers, but pressing keys on another device switched off the modifiers, which is odd, because I can use the RDing pedal like that.

All would be well with a different firmware, I think.

Izzard-UK, 2022-09-29:
> Yeah, that works, key down/up is registered.

I have seen some screenshots of the software so I know that they can be set as a single key press or as a continuous press. What I haven’t seen, yet, is how to set a different keystroke for lifting off the pedal. Hopefully I can use USB Overdrive if the configuration app doesn’t allow it.

Sjokoting, 2025-03-03:
Did anyone manage to reprogram the footswitch?
I curious to know how you managed to enter the bootloader? What did you have to add/modify to the board? I have several hundreds of these. 
Thx in advance.

gizamoluk, 2025-04-19:
@Izzard-UK 
Can you give me a link where I can download  usbsw_config_en.exe?

Izzard-UK, 2025-04-19:
Sorry @gizamoluk - I ditched that pedal and got a decent 3-pedal unit which works with the PC Sensor configuration utility and does what I want. I don't know where to find that exe.

jasonm23, 2025-04-19:
@Izzard-UK @gizamoluk same here.