rgerganov/footswitch #61: Is it possible to disable the key sequence generated by holding the left pedal down for ~7s?

nolanl, 2020-12-01:
I was able to program my 3 pedal switch to generate the keys I'd like (thanks!), but after absentmindedly leaving my foot on the left pedal for around 7 seconds, the footswitch generated Windows-r followed by "http://software.pcsensor.com/pc_en.html", but it did not appear to generate an enter. I assume this is some sort of (undocumented?) functionality to help people get the windows program to program their footswitch, though the URL is currently a 404.

It isn't a big deal, but it would be nice to be able to disable the long-press feature, or even better to reprogram it for a different sequence. I don't have a windows VM to take a USB capture, but if there were unexplained URBs when you reversed the windows program, they might have been for this...

rgerganov, 2020-12-01:
Hm, I don't have this functionality with my switch. I guess this is something they have added in the newer versions.
If anyone finds how to do this, patches are welcome.

moorhexa, 2024-03-15:
I captured this while disabling long-press.
But unfortunatly it still generates the win+r stuff after some seconds.

```
USB URB
    [Source: host]
    [Destination: 4.6.2]
    URB id: 0xffff8908d7b7f540
    URB type: URB_SUBMIT ('S')
    URB transfer type: URB_INTERRUPT (0x01)
    Endpoint: 0x02, Direction: OUT
    Device: 6
    URB bus id: 4
    Device setup request: not relevant ('-')
    Data: present ('\0')
    URB sec: 1710519019
    URB usec: 4072
    URB status: Operation now in progress (-EINPROGRESS) (-115)
    URB length [bytes]: 8
    Data length [bytes]: 8
    [Response in: 647]
    [bInterfaceClass: HID (0x03)]
    Unused Setup Header
    Interval: 8
    Start frame: 0
    Copy of Transfer Flags: 0x00000000
    Number of ISO descriptors: 0

HID Data: 0183080000000000
    0000 0001 = Usage: Pointer: 1
    Padding: 83080000000000
```