rgerganov/footswitch #62: Error

82825264, 2021-01-25:
dokav@dokav-MacBook-Pro footswitch % make
make: pkg-config: Command not found
make: pkg-config: Command not found
cc -Wall -DOSX     footswitch.c common.c debug.c   -o footswitch
footswitch.c:28:10: fatal error: 'hidapi.h' file not found
#include <hidapi.h>
         ^~~~~~~~~~
1 error generated.
make: *** [footswitch] Error 1

FelipeLema, 2022-04-11:
it seems that you need to install [`hidapi`](http://www.signal11.us/oss/hidapi/) first. code won't compile without it

yarmiganosca, 2022-07-15:
I just ran into this issue on a new work laptop, and while I had installed the `libhidapi-dev` package on it (as indicated by the instructions), I hadn't installed the `pkg-config` package yet. Installing `pkg-config` fixed this for me. I do not believe the README needs to include the instructions "make sure you have `pkg-config` installed", since this is probably a rare bug (only happened on a machine I've had for less than a week and am still actively setting up), and because it's pretty easy to read the error message and think "oh hey it's telling me this `pkg-config` command isn't found, maybe I should install that." That said, if the maintainers want to be thorough in listing all the dependencies required, that wouldn't be a *bad* thing, I'm just not sure how many people are going to be running into this problem.

errogaht, 2022-12-14:
`sudo apt install pkg-config` in Ubuntu 22.04 fixed for me

Omustardo, 2023-10-09:
I had the same error message on Debian bookworm. I installed both pkg-config and libusb-dev and then everything worked. From the comment above, perhaps only pkg-config is needed.