rgerganov/footswitch #72: How do I add an emoji to the switch?

edent, 2022-01-26:
```
sudo footswitch -1 -k ☺
Cannot encode key '☺'
```

And
```
sudo footswitch -1 -s '☺'
Cannot encode string: '☺'
```

There's no USB HID key for ☺.

Love this utility - it's so easy to use. Thanks for releasing it.

rgerganov, 2022-01-26:
Unicode symbols (including emoji) are not supported, sorry.

edent, 2022-01-26:
Fair enough! I guess I'll have to find another way of typing 🦶🦶🦶 ;-)

Thanks for the quick response.

closed 2022-01-26 by edent

shervinemami, 2022-01-26:
Hi @edent , each OS or tools will have some ways of mapping a specific keyboard combination to something else. For example, on Linux I use the excellent "AutoKey" application (https://github.com/autokey/autokey). First I tell footswitch to press a rare keyboard combo.

For example, to get my 3-pedal footswitch to have 3 different rare keyboard combos:
`sudo ./footswitch -1 -S '74' -2 -m shift -k F11 -3 -k scrollock               # Open/F14, Shift+F11, ScrollLock`

Then I get AutoKey to always be listening for some of these combos (ie: Create a new "Phrase" in AutoKey and then click on HotKey -> Set, then when it's waiting to learn the hotkey I step my foot on the footpedal to press the rare keyboard combo I configured). Then in the AutoKey phrase or script you can type in emojis or other hotkey combos or run scripts or control the mouse or nearly anything! For emoji's you'll need to change the "Paste Using" mode to use Clipboard instead of Keyboard.

On Windows there is the popular tool "AutoHotKey" that is similar to the Linux "AutoKey" but harder to use.