rgerganov/footswitch #78: Holding Shift, Ctrl etc (Scythe)

Mike-Last, 2022-11-07:
Thanks for the great work on this, the Windows software never detects my Scythe footswitches, so I have to use this to program them.

I would though like to be able to assign a switch to Shift and have it hold shift, while I hold the switch down.  This does not seem to work, I can assign Shift_L but it doesn't seem to have an effect?

rgerganov, 2022-11-07:
Hi, the switches that I have don't support this. They always send `press` and `release` events to the OS. Not sure if newer switches support this.

mlastIDBS, 2022-11-07:
Thanks for the quick reply!

I know the Scythe II can support it as I have previously managed to set it with the windows software (that no longer recognizes the device).  If I set a key to 'a' for example it behaves as if the 'a' key is held down and repeats.

Shift_L though does nothing and I cannot even capture a keypress in the OS, whilst on a normal keyboard I can capture shift just fine.

I have started playing around with HIDAPI on python and can see your software sets the switch to [6, 0, 0, 255, 0, 0, 0, 0] in the feature report, but I don't have a clue what I am doing ;)