rgerganov/footswitch #79: Scythe Footswitch II with different id

Vilhjalmr26, 2023-02-02:
Just got a couple of Scythe USB Single foot switch 2, and a triple, and they all have a new usb id vs the old one I have.  it is now 055a:0998.  Has anyone else gotten one of these?  Have you found a way to use it with this tool?  I like to use the F13 key with my old switch (using this tool) and the official software can't handle that.  Since this program wont currently work for the new one (tried just swapping the IDs and rebuilding, didn't work).  I am a software engineer, but have never reverse engineered something like this, but willing to work with someone if there is any interest in trying to get these working.  Thanks.

rgerganov, 2023-02-02:
I can add support for this Scythe switch if someone donates me the hardware. This is how the current Scythe support was implemented.

Vilhjalmr26, 2023-02-02:
I have a spare one i could send, if i could get it back when you are done, i would ship bith ways obviously.  Where (roughly) are you located so i can determine what shipping cost would be?

rgerganov, 2023-02-02:
Sofia, Bulgaria

shervinemami, 2023-02-02:
@Vilhjalmr26  He mentioned that he'd do the programming & debugging for you if you "donate" (give him) the foot pedal as a gesture of respect for his time & effort. This is common practice in many healthy open-source communities. Whereas for commercial tools you could expect to pay thousands of dollars for this level of custom development.

If you're just lending him the foot pedal temporarily then you're asking for free custom development without even giving him a donation, which isn't respectful. Are you able to give him the pedal or perhaps send him something with it as a gift donation, to show him your respect?

Vilhjalmr26, 2023-02-02:
If shipping costs were going to be super high, i would try to buy from same source i got mine from, have sent to him and hope same version arrived.  Again, sorry for the poor intent conveyed there.

shervinemami, 2023-02-02:
Oh OK good point, yes there's a risk that when you buy 2 of the same footswitch they might have different firmware. So what you say does make sense now that you've explained it!

Vilhjalmr26, 2023-02-03:
Ok, so from looking, shipping wouldn't be terrible, but also the seller I got mine from has a few left, so should in theory be same hw/fw since all of the ones I got from them are the same.  So I guess to you have a preference.  Would you like one of these pedals, or would you prefer to use one for the dev and have some other form of donation for your time and effort?

rgerganov, 2023-02-03:
I would expect that pedals having the same VID:PID would have the exact same firmware. So just make sure that I receive a device with the same VID:PID as yours. Send me an email at rgerganov@gmail.com if you decide to ship me a device and we will arrange the details. Thanks!

rgerganov, 2023-04-20:
I have added support for Scythe 055a:0998 in the master branch. Thanks to @Vilhjalmr26 for donating this switch!

If you find any problems, feel free to open new issues.

closed 2023-04-20 by rgerganov