rgerganov/footswitch #82: Problem programming/using two pedals at the same time

zx485, 2023-05-02:
At first, I like to thank you for your efforts and your great program. It made it possible to avoid Windows :-)

I have two pedals (lying around for some years) and now, I was trying to make them work on Kubuntu 22.04. So I connected both to an old USB-1.0 HUB which should be sufficient for that purpose.

Both are shown in `lsusb`, but I can only program one of them. The other one is kind of ignored. A partial solution was to connect them one after the other to an USB root port and program them one at a time. As a test, I programmed one of them with CTRL-W and the other with CTRL-T.

After reconnecting them to the USB HUB, they both work like programmed, but in `footswitch -r` only one shows as programmed:

    [switch 1]: unconfigured
    [switch 2]: l_ctrl+w
    [switch 3]: unconfigured

The other one works as programmed, but does not show. Maybe its number is higher than three? 
Dunno, so thanks for your input.

Here is my `lsusb` output:

[FootSwitchDevices.txt](https://github.com/rgerganov/footswitch/files/11369336/FootSwitchDevices.txt)

rgerganov, 2023-05-02:
This happens because both pedals have the same vid:pid which is 0c45:7403. Currently there is no way to differentiate between multiple devices with the same vid:pid, sorry.

zx485, 2023-05-02:
That's unfortunate. But I suppose/hope that I won't (have to) change the mappings often.
Thank you for your quick reply and your overall effort.

P.S.: Wouldn't it be possible to append the bus:device ID to the vendor:product ID? This way, at least, the settings would be permanent unless the devices are connected to other ports. Or am I missing something?
 
(You don't have to answer if this would be too complex/breaks something/other.)