rgerganov/footswitch #84: Is there a list of hardware with supported VID:PID?

eslksw, 2023-07-21:
My old pedal just broke and I couldn't get the same one, so I got a new one from PCsensor, but sadly it's not supported. I would like to return the pedal and get a compatible one, since I currently don't have much time for tinkering. Does a compatible hardware list exist? Is there a reliable way to find out the VID:PID of any product? TY in advance.

rgerganov, 2023-07-21:
You can find the list of supported VID:PID in the README

eslksw, 2023-07-21:
Thank you for the quick response. I did see that, but how do I know whether a given pedal has one of the VID:PID listed there? Only `lsusb` comes to mind, but this would require me to buy a pedal, check it with `lsusb` and if not supported, return it and repeat the process. Or maybe I am missing something?