rgerganov/footswitch #96: ctrl + x doesn't work

Schendar, 2024-10-07:
Hi, I'm trying to have my pedals work with guitarix software in AVM distro (debian based).
To change presets i need to send a CTRL - and CTRL + combination to guitarix.
I have programmed the pedals using:
footswitch -1 -m ctrl -k - -2 -s 'd0' -3 -m ctrl -k +
retriving the configuration with footswitch -r gives:
[switch 1]: l_ctrl+-
[switch 2]: d0
[switch 3]: l_ctrl++
but when I press switch 3 it does nothing, while 1 and 2 works perfectly.
Switch 3 works with other combination and CTRL + from keyboard works too.
Can you help me?
Thanks 
Andrea

rgerganov, 2024-10-07:
Try programming the switch with the original software on Windows. If it works, paste the output of `footswitch -r` here.

Schendar, 2024-10-07:
Ok, I've tried with the original software: CTRL + combination is not working. I had to change keyboard layout (I'm italian) to even enter the correct combination but when I memorize it on the pedal and then switch to linux PC, footswitch -r gives "unknown response" for switch#3. So probably is a hardware bug...